beeper/beepy #20: Change Sharp Display driver to support LGVL in monochrome

57Bravo, 2023-07-21:
I was able to get LGVL working on the beepberry, but I had to apply the attached patch to the Sharp display kernel module.

https://lvgl.io/

# LGVL Demo's stitched together
[image]

# Sharp Patch

```
diff --git a/sharp.c b/sharp.c
index 523e5c9..3ee8205 100644
--- a/sharp.c
+++ b/sharp.c
@@ -67,7 +67,7 @@ static struct fb_var_screeninfo vfb_default = {
     .yres =     240,
     .xres_virtual = 400,
     .yres_virtual = 240,
-    .bits_per_pixel = 24,
+    .bits_per_pixel = 8,
     .grayscale = 1,
     .red =      { 0, 8, 0 },
     .green =    { 0, 8, 0 },
@@ -88,7 +88,7 @@ static struct fb_var_screeninfo vfb_default = {
 static struct fb_fix_screeninfo vfb_fix = {
     .id =       "Sharp FB",
     .type =     FB_TYPE_PACKED_PIXELS,
-    .line_length = 1200,
+    .line_length = 400,
     .xpanstep = 0,
     .ypanstep = 0,
     .ywrapstep =    0,
@@ -106,7 +106,7 @@ static struct fb_ops vfb_ops = {
 };
 
 static struct task_struct *thread1;
-static struct task_struct *fpsThread;
+//static struct task_struct *fpsThread;
 static struct task_struct *vcomToggleThread;
 
 static int vfb_mmap(struct fb_info *info,
@@ -212,6 +212,7 @@ int vcomToggleFunction(void* v)
     return 0;
 }
 
+/*
 int fpsThreadFunction(void* v)
 {
     while (!kthread_should_stop()) 
@@ -222,6 +223,7 @@ int fpsThreadFunction(void* v)
     }
     return 0;
 }
+*/
 
 int thread_fn(void* v) 
 {
@@ -273,7 +275,7 @@ int thread_fn(void* v)
             {
                 for(i=0 ; i<8 ; i++ )
                 {
-                    pixel = ioread8((void*)((uintptr_t)info->fix.smem_start + (x*8 + y*400 + i)*3));
+                    pixel = ioread8((void*)((uintptr_t)info->fix.smem_start + (x*8 + y*400 + i)));
 
                     if(pixel)
                     {
@@ -312,7 +314,7 @@ static int sharp_probe(struct spi_device *spi)
 {
     char our_thread[] = "updateScreen";
     char thread_vcom[] = "vcom";
-    char thread_fps[] = "fpsThread";
+//    char thread_fps[] = "fpsThread";
     int retval;
 
 	screen = devm_kzalloc(&spi->dev, sizeof(*screen), GFP_KERNEL);
@@ -332,11 +334,13 @@ static int sharp_probe(struct spi_device *spi)
         wake_up_process(thread1);
     }
 
+/*
     fpsThread = kthread_create(fpsThreadFunction,NULL,thread_fps);
     if((fpsThread))
     {
         wake_up_process(fpsThread);
     }
+*/
 
     vcomToggleThread = kthread_create(vcomToggleFunction,NULL,thread_vcom);
     if((vcomToggleThread))
@@ -404,7 +408,7 @@ static void sharp_remove(struct spi_device *spi)
                 framebuffer_release(info);
         }
 	kthread_stop(thread1);
-	kthread_stop(fpsThread);
+//	kthread_stop(fpsThread);
     kthread_stop(vcomToggleThread);
 	printk(KERN_CRIT "out of screen module");
 	//return 0;
```

xueliu, 2023-08-04:
Good work! it looks very nice !
How many FPS can it achieve ?
Could you share the LVGL project for the Beepberry ? Thanks.

Regards,

Xue

57Bravo, 2023-08-06:
It just uses the main branch. No patches etc.

https://github.com/lvgl/lvgl

I'm not sure I could measure the FPS, but the updating is much more fluid on the live device. That GIF was limited to 1 FPS in my video editing SW to make it smaller for posting.

xueliu, 2023-09-13:
Hi,

I did not make the lvgl running on Beepberry properly.
Can you share your lv_conf.h and lv_drv_conf,h ? Thanks.

57Bravo, 2023-09-14:
Sure. I'll post the configs this weekend when I have access to the machine with the code on it.

57Bravo, 2023-09-19:
```
diff --git a/lv_drv_conf.h b/lv_drv_conf.h
index d40e703..db041b9 100644
--- a/lv_drv_conf.h
+++ b/lv_drv_conf.h
@@ -320,7 +320,7 @@
 #endif
 
 #if USE_FBDEV
-#  define FBDEV_PATH          "/dev/fb0"
+#  define FBDEV_PATH          "/dev/fb1"
 #endif
 
 /*-----------------------------------------
@@ -331,7 +331,7 @@
 #endif
 
 #if USE_BSD_FBDEV
-# define FBDEV_PATH		"/dev/fb0"
+# define FBDEV_PATH		"/dev/fb1"
 #endif
 
 /*-----------------------------------------
@@ -447,7 +447,7 @@
 #endif
 
 #if USE_EVDEV || USE_BSD_EVDEV
-#  define EVDEV_NAME   "/dev/input/event10"        /*You can use the "evtest" Linux tool to get the list of devices and test them*/
+#  define EVDEV_NAME   "/dev/input/event0"        /*You can use the "evtest" Linux tool to get the list of devices and test them*/
 #  define EVDEV_SWAP_AXES         0               /*Swap the x and y axes of the touchscreen*/
 
 #  define EVDEV_CALIBRATE         0               /*Scale and offset the touchscreen coordinates by using maximum and minimum values for each axis*/
```

57Bravo, 2023-09-19:
```
diff --git a/lv_conf.h b/lv_conf.h
index 3137b1a..60dbd74 100644
--- a/lv_conf.h
+++ b/lv_conf.h
@@ -29,8 +29,19 @@ extern uint32_t custom_tick_get(void);
    COLOR SETTINGS
  *====================*/
 
+#define LV_USE_THEME_MONO        1
+#define LV_THEME_DEFAULT_INCLUDE            <stdint.h>      /*Include a header for the init. function*/
+#define LV_THEME_DEFAULT_INIT               lv_theme_mono_init //lv_theme_material_init
+#define LV_THEME_DEFAULT_COLOR_PRIMARY      LV_COLOR_WHITE //lv_color_hex(0x01a2b1)
+#define LV_THEME_DEFAULT_COLOR_SECONDARY    LV_COLOR_BLACK //lv_color_hex(0x44d1b6)
+#define LV_THEME_DEFAULT_FLAG               0 //LV_THEME_MATERIAL_FLAG_LIGHT
+#define LV_THEME_DEFAULT_FONT_SMALL         &lv_font_montserrat_12
+#define LV_THEME_DEFAULT_FONT_NORMAL        &lv_font_montserrat_16
+#define LV_THEME_DEFAULT_FONT_SUBTITLE      &lv_font_montserrat_20
+#define LV_THEME_DEFAULT_FONT_TITLE         &lv_font_montserrat_24
+
 /*Color depth: 1 (1 byte per pixel), 8 (RGB332), 16 (RGB565), 32 (ARGB8888)*/
-#define LV_COLOR_DEPTH 32
+#define LV_COLOR_DEPTH 1
 
 /*Swap the 2 bytes of RGB565 color. Useful if the display has an 8-bit interface (e.g. SPI)*/
 #define LV_COLOR_16_SWAP 0
@@ -147,7 +158,7 @@ extern uint32_t custom_tick_get(void);
 /*Allow dithering the gradients (to achieve visual smooth color gradients on limited color depth display)
  *LV_DITHER_GRADIENT implies allocating one or two more lines of the object's rendering surface
  *The increase in memory consumption is (32 bits * object width) plus 24 bits * object width if using error diffusion */
-#define LV_DITHER_GRADIENT      0
+#define LV_DITHER_GRADIENT      1
 #if LV_DITHER_GRADIENT
     /*Add support for error diffusion dithering.
      *Error diffusion dithering gets a much better visual result, but implies more CPU consumption and memory when drawing.
@@ -539,25 +550,6 @@ extern uint32_t custom_tick_get(void);
  * Themes
  *----------*/
 
-/*A simple, impressive and very complete theme*/
-#define LV_USE_THEME_DEFAULT 1
-#if LV_USE_THEME_DEFAULT
-
-    /*0: Light mode; 1: Dark mode*/
-    #define LV_THEME_DEFAULT_DARK 0
-
-    /*1: Enable grow on press*/
-    #define LV_THEME_DEFAULT_GROW 1
-
-    /*Default transition time in [ms]*/
-    #define LV_THEME_DEFAULT_TRANSITION_TIME 80
-#endif /*LV_USE_THEME_DEFAULT*/
-
-/*A very simple theme that is a good starting point for a custom theme*/
-#define LV_USE_THEME_BASIC 1
-
-/*A theme designed for monochrome displays*/
-#define LV_USE_THEME_MONO 1
 
 /*-----------
  * Layouts
```

xueliu, 2023-09-19:
Hi @57Bravo ,

Thanks for your patch. I would like to know that did you test the program with sharp-drm driver which is currently the default display driver ?

I change my lvgl configuration regarding your patch. But it is still not working. The program shows 4 parallel pictures on the display.

57Bravo, 2023-09-19:
@xueliu - no, you have to use that sharp patch at the top of this thread.

I'll tar my targets src files tonight and post them. I had to modify some of the examples too. You may find them helpful.

57Bravo, 2023-09-20:
@xueliu - Here's the target src directories tar'd up. I didn't clean the builds, so you should clean/rebuild them first.

That "main.c" file in LVGL tar has all of the examples consolidated. Memory restrictions force you to load them in groups.

Sharp Driver
https://drive.google.com/file/d/1UGPaher-F3o9U6Xwb3TCYzIWWgQEXh5m/view?usp=sharing

LVGL
https://drive.google.com/file/d/1mlhjOqSZkcK-yUQX5p__S2dKgbv8E4r6/view?usp=sharing