beeper/beepy #23: Won't install on Bookworm : no raspberrypi-kernel-headers package

danmtripp, 2023-07-28:
x@albiorix  ~/tmp  sudo ./beepberry-setup.sh 
Enabling I2C, SPI, and Console Auto login...
Updating and installing dependencies...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package raspberrypi-kernel-headers
Error: Failed to install dependencies.

 x@albiorix  ~  apt policy raspberrypi-kernel-headers                                    
N: Unable to locate package raspberrypi-kernel-headers

Does this mean I have to go back to buster or bullseye?

danmtripp, 2023-07-28:
Never mind - I guess I'll have to go back to Bullseye - shame - was running my Pi Zero 2W as an ethernet / usb / serial gadget nicely on a 512 GB SDcard...
Going to try out on a 128 GB SDCard (my Beepberry only just arrived this morning)

leoherzog, 2023-10-24:
Also discussed in https://github.com/ardangelo/beepy-ppa/issues/3

> Looks like they're not going to have the same package nomenclature in Bookworm.
>
> https://github.com/raspberrypi/linux/issues/5643
>
> Installation still fails, even with the `linux-headers-rpi-v8` package installed.