beeper/beepy #25: Keyboard not working as expected.

mley, 2023-08-27:
I installed the OS and ran the scripts as mentioned in the readme.

After the final reboot I see the boot messages and the console on the display, but the keyboard seems to be not responding. The beepy-kbd module is loaded: 

```
[    9.604187] beepy-kbd 1-001f: input_probe BBQX0KBD Software version: 0x11
[    9.608168] beepy-kbd 1-001f: input_probe registering input device
[    9.627556] input: beepy-kbd as /devices/platform/soc/3f804000.i2c/i2c-1/1-001f/input/input0
[    9.640042] Character device registered: major = 240, minor = 0
``` 

What's odd is, that when I press the hang-up button for 3 seconds, the character `a` is inputted into the console repeatedly.

dezren39, 2023-08-28:
i would start by ensuring you have a fresh i2c_puppet setup, by repeating the step where you hold the power button while plugging it in so that the file explorer window pops up. ensure you download a copy from the right repo and then drag that in and see what happens when it reboots? If you did already run this, it's worth trying it again.

https://beepy.sqfmi.com/docs/getting-started#firmware-update

Firmware Update

To update the Beepy's firmware:

    Slide the power switch off (left if facing up)
    Connect the Beepy to your computer via USB-C
    While holding the "End Call" key (top right on the keypad), slide the power switch on
    The Beepy will present itself as a USB mass storage device, drag'n'drop the new firmware (*.uf2) into the drive and it will reboot with the new firmware

The latest Beepy firmware can be found here: [i2c_puppet.uf2](https://github.com/sqfmi/i2c_puppet/releases/latest/download/i2c_puppet.uf2)

dezren39, 2023-08-28:
https://github.com/sqfmi/i2c_puppet/releases/tag/v2.1

mley, 2023-08-28:
Updating the firmware was my next planned step. I just did that and the keyboard works now. Thanks for the quick reply.

closed 2023-08-28 by mley