beeper/beepy #30: Screen not displaying anything but static

Cheezeburger23, 2023-09-07:
New beepberry but despite setting up correctly, the screen only displays static.

joeydsmith, 2023-09-11:
same issue for me

OrangeKryptonite, 2023-09-24:
Same issue for me as well

nehafish, 2023-09-24:
Same. I received two of them a few couple-ish weeks ago. A few notes:

- Charged one via USB-C. Only showed static when I turned it on. 
- No other visuals, no reaction to key input. 
- Plugged in my SD Card into Beepberry. Storage Device Not Detected on Mac OS when connected via USB-C.
- I updated firmware via macos + Raspberry Pi Imager. 
- Screen Completely Blank now, but Green Light comes on every time I flip the power switch.
- I imaged the recommended to Raspberry Pi OS to the SD Card on a Digitech SD RW Device.
- Plugged the SD Card back into Beepberry - still no visuals.

Same result with the second, except I did not update firmware.

I'm noticing my macos seems to have detected the Beepberry as a Keyboard device - the desktop winds up responding to key presses when plugged in via USB-C. I don't know if this is normal or otherwise interferring with the device's ability to boot.

arilotter, 2023-09-24:
Follow this exact guide: https://github.com/wildoracle/beepy